StarlingUAS/ProjectStarling #59: Move to upstream MAVROS port

rob-clarke, 2021-04-18:
Currently using a fork for MAVROS ROS2 message support. Is now available upstream on `ros2` branch.

rob-clarke, 2021-06-07:
`mavros_msgs` binary builds are almost in the repos. Switching should speed up image builds a fair amount. `ros1_bridge` is still likely the bottleneck. Should probably wait until `mavros` for ROS2 is more stable before switching wholesale.

rob-clarke, 2021-06-08:
Binaries are in the testing repos now

mhl787156, 2021-09-28:
`mavros` is now available. Question is whether we require the `mavros_extras` package to be ported before moving. A port should be necessary fairly soon as the mavros container is now rather large (>5gb).

mhl787156, 2021-12-07:
`mavros_extras` is now available! Essentially should be ready to port asap
[image]