StarlingUAS/ProjectStarling #91: Custom stack for Aerial Robotics Group Project

arthurrichards77, 2021-09-28:
Add a special docker compose file that launches a single Iris with camera (#88) at Fenswood farm (#89) with a basic scenic representation of the surroundings (#90).

Docker compose only please, as I’d like this to work on Windows.  Please avoid kubernetes.

arthurrichards77, 2021-10-09:
Some progress on this, plus #88 and #90.  Turns out PX4 already ships with a version of Iris with an FPV cam ready to publish to ROS.  The launch file in the zip gets an Iris running in the basic Fenswood scenery with a camera image accessible through ROS.

Key steps:
* Put the ground plane back into the world file, as my drone was just falling through the grass.  Original error was that I'd not changed the `collision` elements of the box when making the grass panels, but for some reason that's not updating.
* Followed [these instructions](https://docs.px4.io/master/en/simulation/ros_interface.html) and then wrote my own custom ROS launch file to run the whole show from ROS.
* Grabbed the Iris model with the cam straight from the SDF file in the directory.  Can't get it to work by choosing it through the vehicle arg.
* Ensured that the launch file sets the Gazebo model path to pick up my grass panels.
* Overriding the world in the launch file to use my Fenswood world.  (Note: paths hardcoded for now as I've not dressed this up in a ROS package.  It's all ROS 1 anyway.

After all that I couldn't fly the thing or connect a ground station!  Just need to learn the ROS interface maybe - but the service callers all got failure responses when I tried to fly.

[aerial_v2.zip](https://github.com/UoBFlightLab/ProjectStarling/files/7316557/aerial_v2.zip)

arthurrichards77, 2021-10-10:
Wondering if the failure to launch is due to lack of a GCS connection.  Had the same problem just using SITL/Gazebo without ROS.  Can’t find the GCS connection when running under ROS wrapper.

rob-clarke, 2021-10-20:
See the repo [here](https://github.com/StarlingUAS/FenswoodScenario) for an initial implementation of this. Currently needs the camera adding. Some stability issues to work out too.

mhl787156, 2021-11-01:
This is solved by the https://github.com/StarlingUAS/FenswoodScenario repository!

closed 2021-11-01 by mhl787156