StarlingUAS/ProjectStarling #94: Dynamic tf_prefix for mavros frame ids

mhl787156, 2021-10-05:
Currently the frame id's for the Transformations given by mavros are hard coded by the single px4 config file (https://github.com/rob-clarke/mavros/blob/master/mavros/launch/px4_config.yaml). This is fine for flying a single vehicle through TF using the [simple offboard](https://github.com/mhl787156/starling_simple_offboard).

However this breaks for multiple vehicles as all of the transformations for the different vehicles clash. A fix has been made from simple offboard [here](https://github.com/mhl787156/starling_simple_offboard/pull/4) which adds `VEHICLE_NAMESPACE` as a prefix to the frame ids which are then broadcast. 

However I do not know how this will interact with the mavros instance which broadcasts (?) its own frame_ids. Therefore a fix is needed to dynamically modify a px4_config.yaml with the correct prefixed frame ids.

mhl787156, 2021-10-05:
@rob-clarke do you think it should be as simple as replacing any `frame_id` or `child_frame_id` with `<vehicle_namespace>/<frame>` (except for the global frames of `map`)? 

In which case my proposal is that in `mavros_setup.sh` we first create a `px4_config_namespaced.yaml` with frame_ids replaced. Then either we: 
1. Provide our own node.launch where we load in the local config params
2. Copy `px4_config_namespaced.yaml` into the mavros installation launch folder to replace the original `px4_config.yaml`

What do you think? 

px4_config.yaml
```
# Common configuration for PX4 autopilot
#
# node:
startup_px4_usb_quirk: true

# --- system plugins ---

# sys_status & sys_time connection options
conn:
  heartbeat_rate: 1.0    # send hertbeat rate in Hertz
  timeout: 10.0          # hertbeat timeout in seconds
  timesync_rate: 10.0    # TIMESYNC rate in Hertz (feature disabled if 0.0)
  system_time_rate: 1.0  # send system time to FCU rate in Hertz (disabled if 0.0)

# sys_status
sys:
  min_voltage: 10.0   # diagnostics min voltage
  disable_diag: false # disable all sys_status diagnostics, except heartbeat

# sys_time
time:
  time_ref_source: "fcu"  # time_reference source
  timesync_mode: MAVLINK
  timesync_avg_alpha: 0.6 # timesync averaging factor

# --- mavros plugins (alphabetical order) ---

# 3dr_radio
tdr_radio:
  low_rssi: 40  # raw rssi lower level for diagnostics

# actuator_control
# None

# command
cmd:
  use_comp_id_system_control: false # quirk for some old FCUs

# dummy
# None

# ftp
# None

# global_position
global_position:
  frame_id: "map"             # origin frame
  child_frame_id: "base_link" # body-fixed frame
  rot_covariance: 99999.0   # covariance for attitude?
  gps_uere: 1.0             # User Equivalent Range Error (UERE) of GPS sensor (m)
  use_relative_alt: true    # use relative altitude for local coordinates
  tf:
    send: false               # send TF?
    frame_id: "map"  # TF frame_id
    global_frame_id: "earth"  # TF earth frame_id
    child_frame_id: "base_link" # TF child_frame_id

# imu_pub
imu:
  frame_id: "base_link"
[... 184 lines cut ...]
  listen_twist: true    # enable listen to twist topic, else listen to vec3d topic
  twist_cov: true       # enable listen to twist with covariance topic

# vibration
vibration:
  frame_id: "base_link"

# wheel_odometry
wheel_odometry:
  count: 2           # number of wheels to compute odometry
  use_rpm: false     # use wheel's RPM instead of cumulative distance to compute odometry
  wheel0: {x: 0.0, y: -0.15, radius: 0.05} # x-, y-offset (m,NED) and radius (m)
  wheel1: {x: 0.0, y:  0.15, radius: 0.05} # x-, y-offset (m,NED) and radius (m)
  send_raw: true              # send wheel's RPM and cumulative distance (~/wheel_odometry/rpm, ~/wheel_odometry/distance)
  send_twist: false           # send geometry_msgs/TwistWithCovarianceStamped instead of nav_msgs/Odometry
  frame_id: "odom"             # origin frame
  child_frame_id: "base_link" # body-fixed frame
  vel_error: 0.1              # wheel velocity measurement error 1-std (m/s)
  tf:
    send: false
    frame_id: "odom"
    child_frame_id: "base_link"

# vim:set ts=2 sw=2 et:
```

rob-clarke, 2021-10-05:
> 1. Provide our own node.launch where we load in the local config params

We should just be able to use the existing launch file and modify the `"config_yaml"` argument. Though at the moment we're using the `px4.launch` file instead so there would be some more more arguments to fill in.

An alternative may be to overwrite the parameters later in the launch file (or in a different launch file), though I'm not sure what the order of execution looks like in ROS2 in that respect, *i.e.* if the parameters would already be loaded by MAVROS before that point.

rob-clarke, 2021-10-05:
Otherwise yes, it should just be replacing the names of the vehicle-specific frames.

mhl787156, 2021-10-06:
> We should just be able to use the existing launch file and modify the "config_yaml" argument

There is also the question of when to do this modification... if it's done in `mavros_setup.sh` we need to work out some mechanism to avoid double writes if a user opens up more than one session (reminder to myself more than anything). 

Make a copy of px4_config and modify the copy the first time mavros_setup.sh is run. Then submit that as `config_yaml` parameter