StarlingUAS/ProjectStarling #147: Foxglove does not plot data

mhl787156, 2022-01-20:
Previous none version of rosbridge suite is v0.1.0. There is problem with the latest version referred to in #146 .

This issue looks at v0.1.0 and tries to identify why foxglove is not plotting any data, but succeeeds at showin data.

An issue has been raised https://github.com/foxglove/studio/issues/2697 but it looks like it might be something from ourside.

mhl787156, 2022-01-20:
Okay, further investigation appears to show that gazebo uses its own times format compared to the rest of ROS. When gazebo is started, it starts its own `/clock` using `rosgraph_msgs/msg/Clock`. This appears to reset all time in the simulation and break how foxglove shows plotting data for everything else. 
Unfortunately not entirely sure how to fix, but have updated the issue with the Foxglove guys so we shall see what they say https://github.com/foxglove/studio/issues/2697

mhl787156, 2022-01-20:
Okay further further investigation seems that this should be solvable through remapping `/clock` into something else, and then remapping it back for mavros on the ros1 side so it doesnt pollute the ros2 list of topics. This may be more annoying when we move to ROS2 but we will cross that line then.

rob-clarke, 2022-01-24:
I feel like these are potentially related:
https://github.com/foxglove/studio/pull/1293
https://github.com/foxglove/studio/pull/2322

closed 2022-06-28 by mhl787156