StarlingUAS/ProjectStarling #157: Fix PX4 SITL Dockerfile

rob-clarke, 2022-02-26:
In upgrading to the latest PX4 version at some point, the non network=host SITL got broken. The sed line [here](https://github.com/StarlingUAS/ProjectStarling/blob/0b5bf28b43fd0027621abe7099b2345636d0c683/simulator/base/px4/sitl.Dockerfile#L31) no longer actually edits the file as it is missing a `-f` before `-m onboard`. This means that the MAVLink output is only ever sent to localhost

mhl787156, 2022-02-27:
Ah I think I have fixed that in the latest dev branch as I ran into that issue: https://github.com/StarlingUAS/ProjectStarling/blob/dev/simulator/base/px4/sitl.Dockerfile

rob-clarke, 2022-02-27:
Yeah that should work. I must have been running latest

rob-clarke, 2022-02-27:
Yeah nightly looks good

closed 2022-02-27 by rob-clarke